eez-open/studio #16: Math operations on imported data

prasimix, 2018-05-22:
Various math operations should be possible on data once imported into ESW. That functionality can be organized in series of plug-ins that can be added over the time from various authors. This is a generic topic that can be break up in many new issues/tickets. Math operation can be performed on all data or selected data between cursors set by user. The following list are an example of possible operations:
 
**Basic operations Y-axis:**

* Max
* Min
* Peak-to-peak
* Average

**Advanced operations Y-axis:**

* [RMS](https://en.wikipedia.org/wiki/Root_mean_square) (root-mean-square)
* Top
* Base
* Amp
* Upper
* Mid
* Lower
* [Overshoot](https://en.wikipedia.org/wiki/Overshoot_(signal))

**Basic operations X-axis:**

* Period
* Frequency
* [Duty cycle](https://en.wikipedia.org/wiki/Duty_cycle) (D)
* Duty cycle (1-D)
* Width (on)
* Width (off)

**Advanced operations X-axis:**

* [Rise time](https://en.wikipedia.org/wiki/Rise_time)
* [Fall time](https://en.wikipedia.org/wiki/Fall_time)
* Delay A to B (between two signals)
* [Phase shift](https://en.wikipedia.org/wiki/Phase_(waves)#Phase_shift) A to B (between two signals)
* A + B
* A - B
* A x B
* A / B
* A || B (logical OR)
* A && B (logical AND)
* Inverse (logical NOT)
* [FFT](https://en.wikipedia.org/wiki/Fast_Fourier_transform)
* [Low pass filter](https://en.wikipedia.org/wiki/Low-pass_filter)
* [High pass filter](https://en.wikipedia.org/wiki/High-pass_filter)
* [Band pass filter](https://en.wikipedia.org/wiki/Band-pass_filter)

**XY:**

* Energy calculations (power over time)

mvladic, 2018-07-12:
Math functions should be implemented as extensions.

prasimix, 2018-08-28:
Partially completed (Max, Min, Peak-to-peak, Average and RMS) to be continued in M3.

prasimix, 2018-09-01:
When ruler is activated, select e.g. 90% of current view. In that case drag operation is changed to graph part selection instead of zoom in.

prasimix, 2018-12-31:
Fixed

closed 2018-12-31 by prasimix