eez-open/studio #9: SCPI commands and queries definitions

prasimix, 2018-05-22:
ESP: Provide extended syntax definition for commands and queries
ESW: Use above mentioned definition when copy selected command or query to command line

Rerouter, 2018-12-02:
http://old.siglentamerica.com/USA_website_2014/Documents/Program_Material/Programming%20Guide.pdf

Page 123 defines how this data should be handled.

Rerouter, 2018-12-26:
Just wondering if there is a formal explanation of how you would like new instruments defined, I've put together about 70% of the siglent SDS1104X-E in the same flavor as the rigol extensions, and the monstrous help documentation, but would like to make sure it all meets spec.

prasimix, 2018-12-26:
We're working on it, step-by-step instructions how to create instrument extension will be included into M3 release

prasimix, 2018-12-31:
Fixed

closed 2018-12-31 by prasimix