eez-open/studio #31: Data from Siglent SSA3021X spectrum analyzer.

dimitarpenev, 2018-06-20:
I have Siglent SSA3021X spectrum analyzer. 
With the general SCPI block I tested that it is recognized and that I can retrieve start frequency, stop frequency and for example trace 1 data.
See attached.

What my action java script will look like if I want just to plot this trace in dBm vs Frequency [Hz] in linear frequency grid and auto-scaling the magnitude?
[image]

prasimix, 2018-06-20:
Thanks, please if possible attach here a trace plot generated by other application, or if not available make a sketch in a picture editor.

dimitarpenev, 2018-06-20:
Sure 
[image]

However i believe  a simple frequency plot starting from f_start ending to f_stop and auto scaled level will be  a good start. Later we may extend it adding few important parameters:
Ref, RBW and detector type

prasimix, 2018-07-05:
We can make a javascript that can in result generate a picture as one that you presented. But, just output of TRAC:DATA? command is not enough. Could you please specify all relevant SCPI commands that we should exchange with your instrument to acquire other parameters (e.g. start and stop frequency, auttenation/magnication, used units, etc.) required to draw usable graph?

dimitarpenev, 2018-07-05:
It is good that you want to start with something more professional!
I am willing to test and provide feedback.

Detailed plot may not be very trivial but lets try.
Some information which can be helpful
1. Documentation with SCPI commands http://old.siglentamerica.com/USA_website_2014/Documents/Program_Material/SSA3000X_ProgrammingGuide_PG0703X_E04A.pdf

2. undocumented command for screen print "scdp" at the result is not interpret as BMP by EEZ (see the attachments)  
[image]

3. Get start frequency in [Hz] "FREQuency:STARt?"
4. Get stop frequency in [Hz] "FREQuency:STOp?"
5. Check the attenuation value in [dB]  "POWer:ATTenuation?"
6. Checking the reference level (top of the screen) in [dBm] "DISPlay:WINDow:TRACe:Y:RLEVel?"
7. Check if the pre amplifier is active [0 or 1] "POWer:GAIN?"
8 Check resolution band width [Hz] "BWIDth?"
9. Check video band width [Hz] "BWIDth:VIDeo?"
10. Detector type [string] "DETector:TRAC1"
11. Read trace 1 data as a vector of [dBm] "TRACe:DATA? 1"
12. how many dB/div we have [dB] "DISPlay:WINDow:TRACe:Y:PDIVision?"

Below I show screenshot and what the above commands return to get some idea
[image]

[image]

prasimix, 2018-07-05:
Please let us know where to find more info about `scdp` command and its output.

prasimix, 2018-07-06:
Could you please provide us what *IDN? query returns. We need that for making instrument extension for that model.

dimitarpenev, 2018-07-06:
06/20/2018 12:49:12 PM
*IDN?
06/20/2018 12:49:12 PM
Siglent Technologies,SSA3032X,SSA3XJA1234567,1.2.9.1

The serial number SSA3XJA1234567 is modified

prasimix, 2018-07-06:
Thanks, that's fine, we just need the name and firmware version.

prasimix, 2018-07-08:
I'd like to know how .bmp generated by `:HCOPy:SDUMp:DATA?` look like that we can insure that it will be displayed correctly.

dimitarpenev, 2018-07-08:
See the attached picture as response of :HCOPy:SDUMp:DATA?
[image]

prasimix, 2018-07-08:
Ok, so this is actually the same as already mentioned `scdp` command?

dimitarpenev, 2018-07-08:
What I remember is that screenshot was not working for this instrument few months ago. 
The author of lxi-tools was working with Siglent and they came up with scdp in their new version. 
Apparent they have enhanced the official ":HCOPy:SDUMp:DATA?" too in the latest firmware I am using now.

prasimix, 2018-07-08:
Good, if you can provide us a .bmp file captured by e.g. lxi-tools or something else could help us to check why it is not already recognized as image file.

dimitarpenev, 2018-07-08:
here it is 
[screenshot1.zip](https://github.com/eez-open/studio/files/2174150/screenshot1.zip)

prasimix, 2018-07-09:
Interesting when bmp file is imported into EEZ Studio it is displayed correctly. 
We need few more info to make instrument extension functional: IP port that you are using for communication and perhaps instrument's USB VID and PID numbers.

dimitarpenev, 2018-07-09:
In EEZ Studio I use Ethernet, port 5025

dimitarpenev, 2018-07-09:
Probably the easiest if you prepare a test EEZ making some intermediate dumps which I run and you analyze?

prasimix, 2018-08-28:
> See the attached picture as response of :HCOPy:SDUMp:DATA?
hcopy_sdump_data

Issue with non-standard query return will be addressed with #9 (scheduled for [M3](https://github.com/eez-open/studio/milestone/2)).

dimitarpenev, 2018-08-28:
What do you mean by "non-standard query return" ?

prasimix, 2018-08-28:
In this case one that doesn't start with # character that follows arbitrary length block data, hence we cannot "interpret" screen capture (i.e. try to display it) as a bitmap.

closed 2022-02-22 by mvladic