eez-open/studio #42: Read USBTMC VID and PID

prasimix, 2018-12-12:
Currently VID and PID have to be entered manually. We have to read it from USBTMC driver and list them in the drop-down list for selection.

Rerouter, 2018-12-15:
Not sure if it helps, but just checked both of my siglent devices, (SDS1204X-E and SAG1021), in case you wanted to know what fields to scrape,

The Device instance path is the device serial number, and the Bus Reported device description is the model number.

SAG1021:
VID "F4ED" , PID "EE3A"
Device Instance Path "SAG10DAQ1RXXXX"
Bus Reported Device Description "SAG1021"

SDS1204X-E:
VID "F4EC" , PID "EE38"
Device Instance Path "SDSMMDBQ2RXXXX"
Bus reported device description "SDS1204X-E"

prasimix, 2018-12-31:
Fixed

closed 2018-12-31 by prasimix