eez-open/studio #78: SCPI Commands to transfer .csv file from Siglent 3021x Spectrum to PC

samarthisrex, 2020-01-30:
I have a .csv file saved in the instrument Siglent 3021x using the following SCPI command-
SocketQuery(s, b'MMEMory:STORe CSV,ABC.csv')

1. I need to transfer it to my local PC connected to it by LAN. I am interacting to the instrument using Socket commands and not pyvisa library. 
2. If I have connected a USB to the instrument how can I save the file 'ABC.csv' to the USB filepath?

prasimix, 2020-01-30:
Please check if that Siglent support upload and try to execute that command from the Studio.

samarthisrex, 2020-01-31:
@prasimix thank you for your response.
I have tried the below command in my python code. 
`SocketQuery(s, b'MMEM:UPL? ABC1.csv')`
I received the following error:
_b'Not found the file = ABC1.csv\n'_

even I tried with the double quotes
`SocketQuery(s, b'MMEM:UPL? "ABC1.csv"')`
Error message:
b'Not found the file = ABC1.csv"\n'

However, when I try the below command 
SocketQuery(s, b'MMEM:UPL? ":ABC1.csv"')
I get below error message with the path from the device
b'Not found the file = /usr/bin/siglent/usr/mass_storage/ABC1.csv"\n'

I can assure you there is a file with the name ABC1.csv

The Code I am using-
[siglent.txt](https://github.com/eez-open/studio/files/4138390/siglent.txt)

prasimix, 2020-01-31:
Perhaps that error message is still not a proof that Siglent support `MMEM:UPL?` and that SCPI parser actually reports that error while parsing received command string. Did you find anywhere in Siglent programming manual how to make file upload? Please also check #31.

samarthisrex, 2020-02-03:
The manual attached does not have any command related to transferring a file from the instrument to PC. However, I was able to transfer the file after mentioning the exact path of the file I intended to move. It was not able to read the file 'ABC.csv' in the default directory of the instrument and I had to write the full path. This is peculiar as for saving the above .csv in the instrument I did not explicitly mention the path. 

[SSA3000X_ProgrammingGuide_PG0703X_E04A.pdf](https://github.com/eez-open/studio/files/4146248/SSA3000X_ProgrammingGuide_PG0703X_E04A.pdf)

closed 2020-02-03 by samarthisrex