eez-open/studio #80: EEZ Studio Project Editor examples

ddavidebor, 2020-04-08:
Hello,
Where can I find examples for EEZ Studio Project Editor?
David

mvladic, 2020-04-09:
Currently, EEZ Studio Project Editor (ESP) servers two purposes:

1) To define GUI for [modular-psu-firmware](https://github.com/eez-open/modular-psu-firmware).

   You can find master project for modular-psu-firmware here: https://github.com/eez-open/modular-psu-firmware/blob/master/modular-psu-firmware.eez-project.

   There is also several projects used by MicroPython scripts which you can find here https://github.com/eez-open/modular-psu-firmware/tree/master/scripts, for example this https://github.com/eez-open/modular-psu-firmware/blob/master/scripts/Curve%20Tracer/Curve%20Tracer.eez-project. These projects use modular-psu-firmware.eez-project as master project (i.e. they use fonts, styles, ... from it).

   Currently, you can't create completely new project for some other development board because all the GUI code is integrated with modular-psu-firmware. But, we have plans to support some of the discovery or nucleo STM32 boards, so it will be possible to create a new GUI project from the scratch.

2) To create so called IEXT extensions, which are description of the instrument (i.e. description od SCPI commands supported by the instrument). This extensions are used in EEZ Studio Workbench (ESW).

    You can find many examples of these in https://github.com/eez-open/studio-extensions. For example this one: https://github.com/eez-open/studio-extensions/tree/master/org/rigol/rigol_ds1000b.

ddavidebor, 2020-04-09:
In my case I am a manufacturer of scientific instrumentation, and would like to build upon the infrastructure provided by the HMI interface, modules, software, firmware, micrpython, scpi support etc etc

prasimix, 2020-04-09:
> In my case I am a manufacturer of scientific instrumentation, and would like to build upon the infrastructure provided by the HMI interface, modules, software, firmware, micrpython, scpi support etc etc

Hi David, please let us know if you need assistance. If you're interested we can try to arrange support on project or consultancy basis.

ddavidebor, 2020-04-09:
I am honestly never going to buy consultancy for a framework of which I am the only user.

closed 2020-06-15 by prasimix