eez-open/studio #86: Does not recognize PNG file from Siglent SDS oscilloscope

vestom, 2020-08-29:
Siglent SDS2000X+ oscilloscope makes a screenshot by using the :PRINt PNG command which returns a binary PNG file. However the PNG file is not recognized as such (see image)

[image]

prasimix, 2020-08-29:
Could you please provide a sample of that binary file acquired by other method (i.e. copy to USB flash, etc.) and attach it zipped here?

vestom, 2020-08-29:
Obtained by using Python pyvisa:
```
    sds.write("PRIN PNG")
    result_str = sds.read_raw()
```

[image]

prasimix, 2020-08-29:
I presume that you can afterwards successfully import and display that file in EEZ Studio by using `Attach file` option?

[image]

vestom, 2020-08-29:
Yes, that works fine :-)
I presume that EEZ Studio should be able to recognize the received binary data as a PNG file itself though?

prasimix, 2020-08-29:
It seems that we have some issue with recognizing file header at the beginning of file transfer. @mvladic has to check that.

vestom, 2020-08-29:
OK. (Just an off-topic question: Is there some documentation for the JavaScript API for making advanced macros? (e.g. like session.addChart())

prasimix, 2020-08-29:
> OK. (Just an off-topic question: Is there some documentation for the JavaScript API for making advanced macros? (e.g. like connection.acquire() and session.addChart())

No, we are missing that as user manual in general :(.
I believe that @mvladic could at least make you a list of what is available/supported for JS.

prasimix, 2020-08-31:
`:PRINt PNG` result cannot be handled properly because that is not valid SCPI query command. Query command has to end with question mark. Therefore its correct format should be `:PRINt? PNG`. One possibility to handle such command is to receive its binary and then retroactively automatically or by adding a new option like "Convert to data block" convert received binary to a PNG image.

prasimix, 2022-06-15:
Closed, due to no response. Reopen if necessary.

closed 2022-06-15 by prasimix