eez-open/studio #94: Screendump in BMP format from Siglent SDS1xxxX-E scope not recognized

fietser28, 2020-11-04:
This issue is different from issue #86 but the solution might be very similar.

The BMP data from the SCDP is not recognised as a bitmap on my Siglent SDS1204X-E scope. Testen both on port 5024 (telnet) or 5025 (sockets).
[image]

Using the following script to download a BMP:
```python
import pyvisa as visa

def main():
  _rm = visa.ResourceManager()
  sds = _rm.open_resource("TCPIP::xx.xx.xx.xx::INSTR")
  print(sds.query("*IDN?"))
  sds.chunk_size = 20*1024*1024 #default value is 20*1024(20k bytes) sds.timeout = 30000 #default value is 2000(2s)
  file_name = "SCDP.bmp" 
  sds.write("SCDP")
  result_str = sds.read_raw()
  f = open(file_name,'wb') 
  f.write(result_str)
  f.flush() 
  f.close()

if __name__=='__main__': 
  main()
```

Results in a BMP file (in zip because of GitHub limitation):
[SCDP.bmp.zip](https://github.com/eez-open/studio/files/5490832/SCDP.bmp.zip)

Uploading this in EEZ-Studio works fine:
[image]

mvladic, 2020-11-05:
In the #86 we added a special query return type called "Non standard data block" to handle such cases. But, in this case, SCDP is not even query so you can't specify return type. It seams that only solution is to use some kind of heuristics to detect that block of data returned by the instrument is bitmap.

[image]

[image]

mvladic, 2020-11-05:
Option added for the SCPI commands called "This command sends back data block":

[image]

You can download latest windows version from here: https://github.com/eez-open/studio/releases/tag/nightly-build. Let me know if you need Linux build.

Also, here is the extension I created for testing: https://github.com/mvladic/studio-experimental-extensions

fietser28, 2020-11-05:
I'm on OS X, did a git pull and a npm build. :-)
Imported the experimental extension you created.

[image]
It works, thank you!

closed 2021-05-11 by fietser28