eez-open/studio #120: EEZ flow - SCPI action doesn't work with multiple inputs

fietser28, 2021-05-09:
If I use multiple inputs in the SCPI action ({a}, {b}) this doesn't seem to work:

Using 1 parameter works:
[image]
[image]

Using 2 parameters doesn't work:
[image]
[image]

mvladic, 2021-05-11:
Fixed!

fietser28, 2021-05-11:
It is working, thanks!

closed 2021-05-11 by fietser28