eez-open/studio #122: EEZ flow - Copy/Paste of actions in flow creates corrupt action

fietser28, 2021-05-11:
1. Right click on an action, select Copy.
2. Move the mouse pointer to another item, richt Click and select Paste
3. Then on top of the action in 1. a new action is placed. Not visually nice, but OK.
4. Move the copied action it to a different location on the page. You know have a ORIGINAL and a NEW action.
5. Connect something to the NEW action. 
7. The line will be drawn to the ORIGNAL action. = not ok.

I assume not everything gets copied during a copy / paste action like this.
If I edit a parameter of the NEW action, it doesn't change in the ORIGINAL action.

mvladic, 2021-05-13:
Fixed.

fietser28, 2021-05-13:
Tested: Works now. Thanks!

Note: the latest commit gave me the following error during build:

> packages/project-editor/core/objectAdapter.ts(52,5): error TS6133: 'createContextMenu' is declared but its value is never read.
> packages/project-editor/flow/flow.tsx(6,5): error TS6133: 'objectsToClipboardData' is declared but its value is never read.

I just commented these two import out, that fixed it.

mvladic, 2021-05-13:
Build errors are now fixed.

closed 2021-05-14 by fietser28