eez-open/studio #124: EEZ Flow - global variables 'binding' not working if variable is deleted/added

fietser28, 2021-05-14:
If I make a flow with a global variable (= defined in the data screen) and use this global variable in an action the following happens:

1. Create the flow, save it. close the flow

2. Open the flow, run it. It runs fine. Behaviour is as expected.
3. Delete the data definition in the data tab
4. Recreate the data in the data tab
5. Run the flow: It doesn't work as expected: Variable is not used properly
6. Save the flow
7. Close the flow
8. Open the flow and run it: It works.

Somehow if you delete/add the variable some bindings are lost.

mvladic, 2021-05-14:
I can't reproduce it. Maybe I'm doing these steps slightly different than you.

fietser28, 2021-05-14:
I also have a hard time to find clear reproducible steps, the above steps don't always work.  It happens occasionally when i add/delete the prevGlobalState variable in the attached flow. When running the lower right chart should be a straight line, if it looks like staircases it doesn't work. Maybe it has to do with where you click/acknowledge the edits in the page. I'm not sure.

[FilterTest.eez-project.zip](https://github.com/eez-open/studio/files/6480593/FilterTest.eez-project.zip)

closed 2023-06-12 by mvladic