eez-open/studio #133: Micropython resource fails if fonts is not enabled

fietser28, 2021-07-25:
If I disable fonts in a V2 resource project (to build a .res file for a MicroPython script) the build fails with the error not very clear error:
"Module build error: TypeError: Cannot read property 'filter' of undefined" 

Tried with recent dev and latest release on OS X.

fietser28, 2021-10-30:
This also applies to EEZ flow applets.

fietser28, 2021-10-30:
It is solved now flow applet (V3) and MicroPython Resource V3 projects.

If I open an old project which has V2 it still gives the error. Should I still be using V2 on new firmwares?

mvladic, 2021-10-30:
V3 is for 1.8, V2 is for 1.7.x. I will fix this also for V2

amossam, 2021-11-15:
On Void linux, Studio 0.9.88, V2 resource project for MicroPyhton, I get same error:
"Module build error: TypeError: Cannot read property 'filter' of undefined"
if I enable only following features: Pages, Actions and Variables.
Project will only build if I also enable: Fonts, Bitmaps and Styles.

When all specified features are enabled, simple project (Hello World) builds and runs on BB3

mvladic, 2021-11-18:
Should be fixed in the latest 0.9.89 build.

prasimix, 2021-11-19:
Fixed

closed 2021-11-19 by prasimix