eez-open/studio #154: Changes to variables in structs passed by reference are not updated in studio

fietser28, 2021-10-30:
For Applet on FW 1.8-alpha2 and git version of studio.

If I create structs and pass the struct as an input to an action, then change one of the variables of the struct in the action this change is not shown in the debugger and screen in studio. It is shown correctly in BB3, so I assume the update is not propagating correctly from BB3 to studio. Don't know if the BB3 firmware or studio is missing this.

Applet:
[image]

BB3 output (correct):
[image]

Screen in studio (incorrect):
[image]

I don't mind the screen is not correctly, but the debug variables on the right are also incorrect.

fietser28, 2021-10-30:
[structin2.eez-project.zip](https://github.com/eez-open/studio/files/7446726/structin2.eez-project.zip)
The used test flow.

mvladic, 2021-10-30:
This should be fixed in the latest Studio & BB3 firmware. 

From this version of Studio it is required to set the type of value in action Input and Output.

[image]

closed 2021-10-30 by prasimix