eez-open/studio #159: Questions about SCPI

sachaw, 2021-12-19:
Hi, I'm in the process of adding support for the Siglent SVA1000X Series instruments,

There is a number of SCPI commands I'd like input on how to handle, commands such as:
- `([:SENSe]:CORRection:CSET[1]|2|3|4:OFF)`
 Currently, I'm creating individual instances for `:SENSe:CORRection:CSET1:OFF`, `:SENSe:CORRection:CSET2:OFF` etc.
- `[:SENSe]:FREQuency:CENTer <freq>`
  Where freq can be either a float value or string suffixed by an enum: `:SENSe:FREQuency:CENTer 0.2 GHz`

Also, Boolean inputs could be represented by a toggle or a checkbox, not a text input accepting a `1` or `0`

I'm sure I'll have some other questions, there's ~800 SCPI commands registered on this instrument!

sachaw, 2021-12-19:
Another thing I've notices how there's a Name and Value column when defining Enums, but only the Name column has any bearing. For example I'd like to be able to have the dropdown for an enum show the name, but the value is pasted when the command is entered: for example, dropdown shows `dBμA` but `DBUA` is entered.

mvladic, 2021-12-19:
For `:SENSe:CORRection:CSET1:OFF`, `:SENSe:CORRection:CSET2:OFF`,  ... you can set `:SENSe:CORRection:CSET[<n>]:OFF`, like this from BB3 instrument definition:

[image]

For `[:SENSe]:FREQuency:CENTer <freq>` just set param type as NR2. NR2 can have unit suffix, but I think it should be without space between number and suffix, i.e. `0.2GHz` is allowed, but no `0.2 GHz`.

I agree for Boolean.

For enums, we should add third column Label and keep Name and Value behavior as it is. Label, if specified, will be show in dropdown instead of Name. You are right that Value is not currently used, but in the future our plan is to use it in the code generation for SCPI commands implementation.

prasimix, 2022-06-15:
Closed, due to no response. Reopen if necessary.

closed 2022-06-15 by prasimix