eez-open/studio #171: Can't Comunicate with RIGOL DP712 over Serial or VISA instrument using Generic SCPI

BillyPocketNC, 2022-02-10:
I have been able to connect to My Rigol DS1054Z over SCPI (Ethernet) and to my DG5072 using Generic SCPI(Also over ethernet)
It seems like I need to change a Serial setting that is available. 

The Rigol DP712 Serial settings are:
BAUD: 9600
Stop Bit: 1
Data Bit: 8
Parity: None

I am able to communicate with the DP712 using the Arduino Serial Monitor;
However I have to set the End of line character to "newline" to get a response. Is there a way to set the end of line character in an instrument definition?

mvladic, 2022-02-10:
Studio already expects '\n' as end of line character.

Download and install this version of the Studio which adds extra logging for serial session:

https://1drv.ms/u/s!AjzKAPRhTu0CovsUiU1N-HzRUGliRw?e=NQgwEI

Close the Studio, and start it again but this time from the command line and send here the output. For example on my computer when I connect to BB3 it looks like this:

[image]

```
C:\Users\mvladic>"C:\Users\mvladic\AppData\Local\Programs\eezstudio\EEZ Studio.exe"

C:\Users\mvladic>
VISA dll Name visa64.dll
(node:1920) Electron: Loading non-context-aware native module in renderer: '\\?\C:\tmp\6e32dd41-f2e4-4ee2-a5e5-a93aa5668a09.tmp.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.
(Use `EEZ Studio --trace-warnings ...` to show where the warning was created)
Creating port, baud_rate: 9600
open
write *IDN?

connected
nextChunk *IDN?

data Envox,EEZ BB3 (STM32),003000373338510D37333535,1.8.alpha3
```

mvladic, 2022-02-10:
Hm, there is no answer from the instrument to "*IDN?\n" command - i.e. this is the line from the my test with BB3 that's missing in your case:

```
data Envox,EEZ BB3 (STM32),003000373338510D37333535,1.8.alpha3
```

Please try once more time with this version:

https://1drv.ms/u/s!AjzKAPRhTu0CovsUiU1N-HzRUGliRw?e=g8O4Z9

mvladic, 2022-02-11:
Great! Let's keep this issue open because we should probably add these options for the Serial connection (along with Baud rate that we already have): data bits,  stop bits, parity and flow control.

mvladic, 2022-04-09:
Fixed!

[image]

closed 2022-04-09 by mvladic