eez-open/studio #174: Timeout (no response to IDN query)

endolith, 2022-03-24:
I am getting this error even though it looks like it's responding to me:

[image]

prasimix, 2022-03-24:
If you're using USBTMC interface on Rigol then you can expect such error. Try Ethernet or VISA.

prasimix, 2022-03-24:
See also: https://www.envox.eu/how-to/how-to-capture-a-screen-shot-from-an-rigol-dso-and-many-others/

endolith, 2022-03-25:
> If you're using USBTMC interface on Rigol 

I'm not sure, I'm using this:

prasimix, 2022-03-25:
That is VISA, it should work correctly with R&S VISA installed.

endolith, 2022-03-25:
Can you give me any more hints?  

https://www.envox.eu/how-to/how-to-capture-a-screen-shot-from-an-rigol-dso-and-many-others/ says nothing about this except 

> EEZ Studio has been tested with R&S®VISA

After installing `RS_VISA_Setup_Win_5_12_3.exe` there is another option:

[image]

But both options produce the same "no response to IDN" error, despite obviously receiving a response to the IDN:

[image]

In the R&S Configure tool:

[image]

I can create a new Resource, but it has the same string:

[image]

And still has the same confusing error

prasimix, 2022-03-26:
If the R&S tool cannot establish a connection with the device, then EEZ Studio will not be able to do so via the VISA interface. The problem is possible on the Rigol side, or it can be up to the operating system, for example Linux has some problems with USBTMC with newer USB host controllers due to which it is not possible to establish a connection. If that model comes with Ethernet, you can try that interface.

endolith, 2022-03-29:
It is establishing a connection, though?  It sends IDN and the scope responds, no?

prasimix, 2022-03-29:
Yes, but if you're on Linux and using USBTMC then due to some buffer issue it cannot keep connection alive. Check this on the following [link](https://www.teuniz.net/DSRemote/):

[image]

endolith, 2022-03-29:
I'm using Windows 10 Version	10.0.19044 Build 19044 and the USB VISA interface ![
](https://user-images.githubusercontent.com/58611/160043862-00e71685-44c8-4572-b4dd-aeeb40a08b62.png)

prasimix, 2022-03-29:
Ok, I see. Don't know if that model has Ethernet interface to try it out, since that is all that I can suggest in this moment.

prasimix, 2022-03-29:
In general, if you cannot establish connection with R&S tool you cannot expect that EEZ Studio via VISA interface works any better.

closed 2022-06-15 by prasimix