eez-open/studio #194: Dashboards won't run/debug on OS X with 0.9.93

fietser28, 2022-08-01:
If I create a new dashboard in 0.9.93 (hello world text) and run it or debug it (adding a start  and constant with a breakpoint), nothing happens at all in both cases.

The debug console of studio shows:

[image]

Upon the start of studio the debug windows shows:

[image]
(the yellowish messages continue for a while)

The app.asar file is present:
[image]

Running OS X 12.15 (latest) on an M1 Pro CPU. Tested both the DMG and ZIP version, no difference. 
I think this also happened with 0.9.92 but didn't validate.

fietser28, 2022-08-01:
If I press run the following exception appears (same as above but extended, giving more info):
[image]

mvladic, 2022-08-13:
Fixed in the latest 0.9.94 build.

fietser28, 2022-08-14:
Works now. Thanks.

closed 2022-08-14 by fietser28