eez-open/studio #196: Rounding of float behaviour in text labels is inconsistent

fietser28, 2022-08-14:
Using Math.round() in a text label doesn't work as I would expect. The behaviour between double and float is also different.

Example flow:
[image]

Will result in (after final breakpoint):
[image]

The B row float seems out of oder. The other values look ok to me.

Using 0.9.94 on Mac M1.

Project file:
[RoundTest.eez-project.zip](https://github.com/eez-open/studio/files/9334075/RoundTest.eez-project.zip)

fietser28, 2022-08-15:
I encountered a possibly similar/same problem in the SCPI component/commands:

If I send a float to the SCPI command with a value of 0.25 then 'inf' is send to the BB3 (according to the debug log). I seems some values like 0, 1, 2 work but with some fractions not. I can't find a pattern in this.

My workaround is similar as with the text component (setval is a string):
[image]

mvladic, 2022-08-15:
Please test the latest build and let me know if `'inf'` bug is fixed.

mvladic, 2022-08-15:
We have `float` and `double` type because for embedded use `double` could be too expensive performance wise. General advice for dashboards is to use `double` to prevent any rounding errors when converting `float` to `double`, because `double` (i.e. number) is only type that JavaScript knows.

Let me expain why `Math.round(offset, 2)` differs from `Math.round(offset, 2) + ""`. These two expressions are evaluated inside engine (engine knows how to work with both `float` and `double`!). Former expresssion results in value of `float` type, and later result in value of string type (properly rounded to two decimal digits inside engine and converted to string value for use inside JavaScript). When result value of former expression is converted first from engine `float` value to Javascript `double` and then from `double` to `string` we can see that it is not rounded as expected because conversion from `float` to `double` adds these decimal digits:

```
0.33000001311302185
         **********
```

So, again, to prevent these rounding errors always use double in dashboard projects.

fietser28, 2022-08-15:
> Please test the latest build and let me know if `'inf'` bug is fixed.

Tested it, it seems fixed. Thank you.

fietser28, 2022-08-15:
Clear explanation. I also realise I'm not using (abusing) Math.round for numerical purposes but for displaying/formatting purposes. Is there a better alternative e.q.  sprintf("%2.2f", float) to control the float to string conversion?

mvladic, 2022-08-17:
Yes, we should probably make string formatting function `String.format(...)` similar to `sprintf`.

closed 2023-06-12 by mvladic