eez-open/studio #197: EEZ flow: In Plot: Line  labels not shown in run

fietser28, 2022-08-15:
In Run or Debug mode/window. Trace 0 - 4 is shown instead of the line labels:
[image]

In Edit mode/window (these are the set names of the lines):
[image]

A dashboard project using 0.9.94 on Mac M1.

mvladic, 2022-08-15:
Can't reproduce. Could you please attach a simplified dashboard project with this problem?

Here is my testing dashboard project:

https://1drv.ms/u/s!AjzKAPRhTu0CpLR7aGmp3A80cslLRQ?e=8pcuZC

Here is the screenshot in Run mode:

[image]

fietser28, 2022-08-15:
I've modified your project file (attached in this message) to produce the error:
I think it has to do with the reset input:  The first time you press the start button  the labels are wrong. On subsequent presses the labels are correct.
I've also played with the X value but that doesn't seem to make any difference.

[test_line_chart_w_reset.eez-project.zip](https://github.com/eez-open/studio/files/9339230/test_line_chart_w_reset.eez-project.zip)

[image]
[image]

mvladic, 2022-08-17:
Should be fixed in latest build (0.9.95).

fietser28, 2022-08-19:
The previous example is working with today's build.

The attached example is not (anymore?):
- With a single trace the label is not shown
- Data.now() doesn't plot any data anymore. A constant or variable works as expected.

[PlotTest.eez-project.zip](https://github.com/eez-open/studio/files/9385206/PlotTest.eez-project.zip)

mvladic, 2022-08-21:
I suppose you mean `Date.now()`, not `Data.now()`? 

I can confirm that first point doesn't work, but second point I can't repeat since I can see date's on X axis:

[image]

mvladic, 2022-08-21:
It appears that Plotly library chooses not to show legend if there is only a single trace line. Luckily, we can override this behavior if we set its `showlegend` property to `true`. I added a new property to LineChart widget called "Show legend", by default it is set to `true`. If set to `false` it will not only hide legend for single trace charts, but for all charts.

[image]

fietser28, 2022-08-21:
All fine now. Thank you.

closed 2022-08-21 by fietser28