eez-open/studio #219: Initial support for RP2040

goran-mahovlic, 2022-11-20:
https://github.com/cnkoala/rp2040-eSPI-lvgl

Also this screen looks nice as a target

https://www.waveshare.com/product/pico-restouch-lcd-2.8.htm

fietser28, 2022-11-20:
The example above uses platformio + standard arduino. This has the advantage that a lot of libraries are available (Arduino)
As far as I know the default arduino platform only uses 1 core of the RP2040.

I think flow requires something like FreeRTOS. I know of 2 ways to do that on a RP2040:
- Plain: Pico SDK + FreeRTOS, library support may be challenging.
- https://github.com/earlephilhower/arduino-pico. This combines arduino + FreeRTOS on a RP2040. It is also supported by a development platform of platformio: https://github.com/maxgerhardt/platform-raspberrypi.

My experience with the R2P2040 so far has been rocky due to all the software platforms/libraries being new on the RP2040.

goran-mahovlic, 2022-11-23:
I am started here, but do not know how much Arduino links things so I am ending in errors.

But I think there is not much to do to have support...

https://github.com/goran-mahovlic/arduino-lvgl-eez-demo

goran-mahovlic, 2023-05-25:
We have platformio version https://github.com/eez-open/raspberrypi-pico-tft-eez so I will close this issue

closed 2023-05-25 by goran-mahovlic