eez-open/studio #221: Easier monitoring and changing basic settings

mattico, 2022-11-30:
Many commercial instruments remote control capability just copies the physical interface of the device. Often they just show the device's screen and the control buttons are more-or-less equivalent to the physical buttons on the device. Not a great interface to use with a mouse, but it works. Often the scripting capability is limited, you have to do it yourself with GPIB.

EEZ Studio has gone in the opposite direction. Most of the effort seems to be going into advanced scripting capabilities, GUI editors, etc. This makes sense, as those capabilities are what make the EEZ hardware special. However, most of the time I just use my BB3 as a power supply, and I think that's pretty common. I've been baffled for a long time that EEZ Studio doesn't have any view that just shows what voltage and current each channel has, and lets you change the setpoints. It hasn't been a huge bother because I could use the BB3's screen; it's not always very convenient when the workbench is crowded, but it's fine. Recently I've been working remotely, and have had to use EEZ Studio more to do basic things like check voltage and current and change the values, and it's become more irritating.

I know all of those things can be controlled with SCPI commands, shortcuts, and scripts. But I shouldn't have to, and the ease of use and visual presentation will never match a specifically designed interface. I find myself spamming the screenshot button in EEZ Studio, because the BB3's interface is really good at communicating the state of the device, and EEZ Studio's interface is not.

I don't intend for the tone to be overly negative, I appreciate all the work that's been done on EEZ Studio. But I would appreciate some attention paid to basic functionality along with advanced scripting.

This may be a dupe of https://github.com/eez-open/studio/issues/113, I'm not exactly sure what that issue is talking about.

prasimix, 2022-11-30:
Thanks for the feedback. We are well aware of this, but so far we have not been in a position to make a remote console what many have had for a long time. #113 is a kind of reminder that we need something like that. (the task has yet to get a description besides the title itself).

prasimix, 2023-07-01:
Task #113 is closed, so I'm closing this task as well. See also https://github.com/eez-open/bb3-dashboard-app

closed 2023-07-01 by prasimix