eez-open/studio #223: LVGL Font preview is scrambled (on OS X)

fietser28, 2022-12-07:
Minor issue: On latest commit font preview in LVGL project looks scrambled:
[image]

The fonts are fine in editor, run, debug and on device itself. Only the preview is scrambled.

Of course it can also be just my build on OS X on M1.

mvladic, 2022-12-08:
It works on Windows. Do you have an M1 or M2 Mac, i.e. did you build the ARM binary?

mvladic, 2022-12-08:
BTW I tried but I can't build a functional ARM64 binary using github actions. I noticed that X86 binary is really slow.

fietser28, 2022-12-08:
I have an M1 Pro, so ARM processor. I build native on arm. 

Latest commit of today ([lv_img_conv added](https://github.com/eez-open/studio/commit/19d7514b2ef571b693af10140b07e16dfcc59d79) )didn't result In a working studio so I reverted back to [improvements in animation timeline keyframe control points](https://github.com/eez-open/studio/commit/95da068dec5389635dc13d65643d6eef12ca0482) AND I had to revert back to node v16 (This is not unique to studio). That results in a working studio.

The latest images on GitHub don't work on my M1: Studio starts but page stays blank.

mvladic, 2022-12-09:
The version was broken on Windows also. It is fixed now.

fietser28, 2022-12-09:
Build is working now (tested not. Font previews are still scrambled, they are working in editor, run and on mcu.

mvladic, 2022-12-27:
I can't reproduce this problem on M1 Pro. Can you share the details which font file (TTF) you are using?

fietser28, 2022-12-27:
https://www.1001fonts.com/digital-7-font.html
Both the digital-7-regular and the digital-7-mono

mvladic, 2022-12-27:
Still, can't reproduce. I tried both fonts files with x86 and arm64 version and with various font sizes.

fietser28, 2022-12-27:
Probably something with my build. It's also not an important bug, just cosmetic.

fietser28, 2022-12-27:
Narrowed it doen: If I change the bits per pixel (bpp) to 4 then the fonts are scrambled. The default bpp of value 8 works fine.

mvladic, 2022-12-27:
Fixed

fietser28, 2022-12-28:
Confirmed, thanks.

closed 2022-12-28 by fietser28