eez-open/studio #225: LVGL widget TextArea doesn't accept all forms of expressions

fietser28, 2022-12-09:
When setting the Text of a TextArea widget to a variable it works.

When setting the Text of a TextArea to an expression (e.g.   my float + "U") both in studio (with error: "Failed to assign Text in Textarea widget: (null)") and on MCU (ui task freezes).

Not working:
[image]

Works:
[image]

Also working (this is a label not a texture):
[image]

mvladic, 2022-12-10:
Fixed.

fietser28, 2022-12-10:
Working. Thank you.

closed 2022-12-10 by fietser28