eez-open/studio #228: flow: optionally export enum and struct definitions

fietser28, 2022-12-26:
Please add the option to export the enum and structs defined in a studio to an appropriate .h file (vars.h?)
This allows for a more consistent integration between flow and code.

mvladic, 2023-05-26:
In the next version of EEZ Studio there will be a better support for enums in LVGL projects.

For example, if you have following enum definition in a project:

[image]

and following native variable definition in a project:

[image]

you can implement get and set native functions for this variable.

First, you must enable in `vars.h` file template option to generate enum definitions:

[image]

This will be already enabled for the new projects.

Now, in `vars.c` you can implement get and set functions like this:

```
MyEnum my_enum_var = MyEnum_Member2;

extern MyEnum get_var_my_enum_var() { return my_enum_var; }
extern void set_var_my_enum_var(MyEnum value) { my_enum_var = value; }
```

mvladic, 2023-05-26:
In the next version of EEZ Studio there will be a better support for structs in LVGL projects. First of all, struct variable can't be a native variable, it can only be a flow variable. But, it will be possible to access and modify these variables from the native code.

For example, let say you have following struct definition in a project:

[image]

and following global variable definition in a project:

[image]

First, you must include following in vars.h (it will be already include for the new project):

[image]

And you must have a new file template called structs.h (it will be already include for the new project):

```
#ifndef EEZ_LVGL_UI_STRUCTS_H
#define EEZ_LVGL_UI_STRUCTS_H

#include <eez/flow/flow.h>
#include <stdint.h>
#include <stdbool.h>

#include "vars.h"

using namespace eez;

//${eez-studio FLOW_STRUCTS}

//${eez-studio FLOW_STRUCT_VALUES}

#endif /*EEZ_LVGL_UI_STRUCTS_H*/
```

Now, you can access and modify struct variable like this:

```
#include "vars.h"
#include "structs.h"

// in this native action we will access the field values of options struct variable
void action_options_from_flow_to_native(lv_event_t* e) {
  // get options global variable as OptionsValue object which can be used to get/set field values
  OptionsValue options = flow::getGlobalVariable(FLOW_GLOBAL_VARIABLE_OPTIONS);

  // get switch_option field (boolean)
  bool switch_option = options.switch_option();
  
  // get num_options_arr (integer array)
  auto num_options_arr = options.num_options_arr();
  // iterate over array values
  for (unsigined int i = 0; i < num_options_arr.length(); i++) {
    // get i-th value in array
    int value = num_options_arr.at(i);
  }

  // get my_enum_field field (enum MyEnum)
  auto my_enum_field = options.my_enum_field();
}

// in this native action we will modify the field values of options struct variable
void action_options_from_native_to_flow(lv_event_t* e) {
  OptionsValue options = flow::getGlobalVariable(FLOW_GLOBAL_VARIABLE_OPTIONS);

  // modify switch_option field
  options.switch_option(false);

    // set "num_options_arr" field to [10, 20, 30]
    ArrayOfInteger arr(3);
    arr.at(0, 10);
    arr.at(1, 20);
    arr.at(2, 30);
    options.num_options_arr(arr);

  // modify my_enum_field
  options.my_enum_field(MyEnum_Member2);
}

```

fietser28, 2023-06-12:
If I try to include the structs.h in my project I get a lot of errors:

[image]

I checked al the necessary statements in the .h files and they look fine. I can't find where the StringValue/FloatValue should be defined.

I have a struct defined as:
[image]

Am I doing something wrong?

fietser28, 2023-06-13:
That fixed it. thank.