eez-open/studio #231: Flow: comment/description box size is off for small components

fietser28, 2022-12-27:
just cosmetic:

[image]

The yellow description boxed appear to have a later large minimum width.

mvladic, 2022-12-27:
It's intentional - the minimum size is 200px

mvladic, 2022-12-27:
Minimum size exists so it doesn't wrap to much longer descriptions. Maybe it should be centered with the widget so it doesn't look off.

mvladic, 2022-12-27:
For example, not centered:

[image]

Centered:

[image]

fietser28, 2022-12-27:
Centered looks good for larger text.

For small texts (text width is a single line and less then width of component) I would make it the width of the widget.
In that way there is no 'empty box' sticking out.

mvladic, 2022-12-27:
I don't how to do in CSS: remove min-width if smaller text. But, maybe I should remove `min-width: 200px`. In that case your example will look like this:

[image]

fietser28, 2022-12-27:
This looks better to me, but that is personal taste. You probably should keep a small minimum.

mvladic, 2022-12-27:
This is already at minimum defined for the action component.

mvladic, 2022-12-28:
I removed the "min-width" of 200px for comments. It will always be the same width as the component the comment refers to. "min-width" for the component is 87px (don't ask me how I got that number :) ).

fietser28, 2022-12-28:
Looks good now to me. Thanks.

closed 2022-12-28 by fietser28