eez-open/studio #235: Flow: Watch action causes UI freeze on MCU

fietser28, 2023-01-01:
Watch action works fine in studio run/debug but not when running om MCU (RP2040+TFT)

Simple flow:
[image]

I've setup a project to demo this:
https://github.com/fietser28/eez-flow-watchtest

The demo doesn't work, but if you remove the watch action it starts working. It looks like the initial display setup works (layout is shown) but all the text fields are NOT filled correctly.

mvladic, 2023-01-01:
Looks like `watchtest.eez-project` is not the latest version, can you push the latest version?

mvladic, 2023-01-01:
Are you sure that `ui_tick()` has been called periodically?

fietser28, 2023-01-01:
1. pushed latest version.
2. Yes it is, if I just remove the watch action the display updates nicely every second.

mvladic, 2023-01-01:
I can't reproduce this problem. What you mean by this: "... all the text fields are NOT filled correctly."?

mvladic, 2023-01-01:
I tested with web simulator repository (https://github.com/mvladic/lv_web_emscripten):

[image]

mvladic, 2023-01-01:
BTW Watch is not required to update label text as expressions is evaluated all the time in `ui_tick` function and label text is updated if text expression result is different then current label text.

fietser28, 2023-01-01:
I used this watch as a simple example (ran into it in more complex case and tried to narrow down the problem).

In studio the simulation it also works fine but on RP2040 it doesn't:

I'v added an extra project file to the GitHub (https://github.com/fietser28/eez-flow-watchtest)  with only the change that makes the UI freeze (probably on startup or first tick run?).

The watchtest project results in blinking and wd var label alternating true/false:
[image]

The watchtest-notworking project results in the picture below, without any UI updates. It looks like the UI task freezes.
[image]

I've seen this earlier in other cases, always on RP2040 so it might also be a problem in the arduino framework/toolchain/freertos but given the fact is was always with the watch action it seems unlikely. In these cases removing the watch action and using variables directly solved the problem.

mvladic, 2023-01-02:
It also works on M5Stack, so it must be something with RP2040/Arduino. If you can debug then put the breakpoint at line 54 in eez/flow/hooks.cpp:

[image]

and paste here the call stack.

fietser28, 2023-01-02:
I did some stepping/breakpoints in the debugger and it looks the ui_tick() is entered but never leaves:
[image]

This looked especially strange to me because of the last if / break statement: This should timeout, but it doesn't. Further stepping revealed millis() is overloaded with lv_tick_get() in os.cpp:
[image]

This is not working because my display loop looks like:
[image]
meaning lv_tick_inc never gets called when ui_tick is still processing.

In different code I placed  lv_tick_inc in a separate timer loop, but placed a mutex around all lv* and ui* calls because the rp2040 is multi-core and multi-thread. This didn't avoid the problem, so I updated the code in https://github.com/fietser28/eez-flow-watchtest without a mutex (lv_tick_inc is safe irt the other calls especially since lv_tick_get has a poor mans mutex test).

Conclusion: put lv_tick_inc in a separate task and it works. If this is intentional/by design we can close this problem.

mvladic, 2023-01-02:
What you mean by this "**If this is intentional/by design** we can close this problem."?

mvladic, 2023-01-02:
I wonder if maybe we should add another #ifdef for ARDUINO and in that case call `::millis()` which exists on that platform.

prasimix, 2023-05-09:
@fietser28 please let us know if we can close this ticket (it will be included in the next milestone)

fietser28, 2023-05-10:
Watch actions are now working fine in my case. It can be closed.

For the record: I currently use the following setup with Arduino + FreeRTOS:
```c
static void guiTimerFunction(TimerHandle_t tm)
{
  lv_tick_inc(MY_LV_TICK_TIME);
}

void gui_task_init(void)
{

  guiTimerHandle = xTimerCreate("", pdMS_TO_TICKS(MY_LV_TICK_TIME), pdTRUE, (void *) 0, guiTimerFunction);
  xTimerStart(guiTimerHandle, 10);

  xTaskCreate(guiTask, "", 4096 * 4, NULL, TASK_PRIORITY_UI, &guiTaskHandle);
};
```
And in the main guiTask the following l have:
```c
while (1)
  {
    lv_task_handler();
    ui_tick();
    vTaskDelay(pdMS_TO_TICKS(MY_LV_UPDATE_TIME));
  }
```

closed 2023-05-10 by prasimix