eez-open/studio #244: can find device "DS1000Z series", but connect button is greyed out

crNewton, 2023-02-10:
EEZ Studio: V0.9.89
Scope: RIGOL DS1104Z+

I've been using EEZ Studio with my scope, but today I couldn't connect. 
I'm connecting using the USB cable that came with the scope, and worked before. 
Tried various USB ports & tried a reboot.

My scope is discovered as 'device', but the 'connect' button is greyed out,
Shown in the screen capture below (or [here](https://imgur.com/a/Ry554tm.gif))

I doesn't look like a driver issue.
Any suggestions how to pinpoint the problem, or a solution?
[image]

mvladic, 2023-02-10:
What is VID and PID for that device?

crNewton, 2023-02-10:
> What is VID and PID for that device?

VID_1AB1&PID_04CE

mvladic, 2023-02-10:
I think I found what the problem is. Due to a bug, when your device is the first in the list, the Connect button will be disabled. I just committed a fix. If you are willing to test it, I can add you as collaborator to this repository so you can download DRAFT releases.

crNewton, 2023-02-13:
> I think I found what the problem is. Due to a bug, when your device is the first in the list, the Connect button will be disabled. I just committed a fix. If you are willing to test it, I can add you as collaborator to this repository so you can download DRAFT releases.

Fast find, thank you! With V0.9.89 I found your described behavior. 
I can't fully test it, I can't get my scope to be the first device in the list anymore .

closed 2023-02-13 by crNewton