eez-open/studio #252: (LVGL) EEZ flow: edit font definition in UI

fietser28, 2023-05-07:
Use case: Using a cusom font I wanted to add a single symbol I didn't select during the initial definition in studio.

Currently it is possible to create a new font definition but it is not possible to alter the definition afterwards. 
This is especially annoying when using ranges and symbols: E.g. adding a single symbol is currently not possible.

The information panel currently doesn't show which ranges/symbols are used to create the font. That would be nice to.

Another issue I noticed: It seems the ttf font is included in the eez-project file. Although this is convenient, it also is a risk: A lot of fonts have specific copyrights. I think the inclusion should be an option and if the user can edit the font definition project can be distributed without fonts included and the user can provide/choose there own.

mvladic, 2023-05-07:
Adding single symbol is possible when the new font is created. For example, you can enter in Ranges field: 32-127,176,182,200-210

mvladic, 2023-06-10:
I fixed the first two issues:

- I added edit icon in the characters list panel:

[image]

This will open up the form to edit Ranges and Symbols:

[image]

- In the "information panel" it is now shown Ranges and Symbols:

[image]

For now, I would not do anything about the third issue. There is large selection of free fonts, for example from google. If someone still wants to use copyrighted font he should modify project file before publicly sharing it. If he keeps the project file private, for example inside private github repository, than there is no issue I think.

fietser28, 2023-06-12:
First two are working. Thank you.