eez-open/studio #257: EEZ Flow: Styles can be hierarchical/derived.

fietser28, 2023-05-13:
Working with styles is pretty essential in EEZ Flow if the project get bigger. To make it more manageable I would like it to be possible to derive styles from each other in the UI (it doesn't have to be in the resulting code): 

- [ ] A style can be derived from another style so you only manage a small amount of differences and are not copying the style and making adjustments.

- [ ] It would be nice if this is also shown visually in the style browser (tab).

mvladic, 2023-06-10:
Done.

Now, you can make hierarchy of styles by using drag and drop (similar to Widgets Structure panel):

[image]

fietser28, 2023-06-11:
If I drag and drop small_label_symbol under small_label:
[image]
[image]

Small label symbol only has a different font, the color is the same. It doesn't remove the changes not needed anymore (no problem, allows for manual verification) AND if I unmark the color of small label symbol then I would except the color to be inherited from small label, but it doesn't (I would call this a bug).

But: Now closing the project file and reopening:
[image]
It behaves as expected.

After this I tried it again with a different combination and was not able to reproduce...

mvladic, 2023-06-12:
This problem can be easily reproduced:

[image]

mvladic, 2023-06-12:
This is fixed now.

mvladic, 2023-06-12:
> ... Small label symbol only has a different font, the color is the same. It doesn't remove the changes not needed anymore (no problem, allows for manual verification) ...

I'm not sure I want to *automatically* remove changes that are not needed anymore. Because, if you first move style at some  position and if you change your mind and move somewhere else then it can loose modification. Maybe we should extend this menu:

[image]

With another option called "Remove Unnecessary Modifications".

mvladic, 2023-06-12:
Hm, we don't have this menu in style properties, only in widget style properties.

fietser28, 2023-06-12:
> This problem can be easily reproduced:
> 
> [image]

With automatically I mean: drag a style to become a 'child' style AND both these styles have a setting explicitly set to the same value it will remove this setting in the child style because it is already set in the parent child. Doing this automatically after the dragging doesn't seem like a good idea, but implement it like 'reset all modifications' but in the styles menu and calling it 'Remove redundant modifications' or 'Simplify style'.

I really like this enhancement as it is already implemented. It allows for good overview and cleanup of styles (in combination with the 'find all references' feature) without needing a 'big design up-front'.

fietser28, 2023-06-12:
> Hm, we don't have this menu in style properties, only in widget style properties.

I think this is exactly what is described above.

mvladic, 2023-06-13:
I added "Remove Redundant Modifications" option in context menu for the style.

[image]