eez-open/studio #264: Add MQTT support as Extension

prasimix, 2023-05-28:


fietser28, 2023-05-29:
This is for flow running in studio (dashboards) or also for embedded (e.g. LVGL flow)? 

I think both have a valid use case, but the implementations are different. The embedded MQTT libraries vary per platform so a generic implementation might be difficult.

mvladic, 2023-05-29:
This should be for all project types (Dashboard, EEZ-GUI and LVGL), i.e. all project types will have the same action components for MQTT. But, implementation will be provided only for Dashboard. For EEZ-GUI and LVGL there will be some interface defined (bunch of native functions) which must be implemented by the user for the target platform.

prasimix, 2023-05-29:
Since we have MQTT implemented on BB3 we can create an example project that uses the same code but on a Discovery board that we already have.

prasimix, 2023-06-30:
Fixed

closed 2023-06-30 by prasimix