eez-open/studio #267: SCPI terminal: add option to link the SCPI command input to the SCPI documentation search.

fietser28, 2023-06-04:
When using the SCPI terminal in studio I frequently use the feature to lookup SCPI commands in the 'documentation browser'. I need to frequently switch between the SCPI input line and the documentation search field (using mouse).

I think it would be very nice to have a checkbox (above the search box of the documentation browser or a link icon in the search box?) to link the command input with the documentation browser: If this checkbox is on then the SCPI documentation search dynamically follows my command input:

[image]

mvladic, 2023-06-04:
Done.

fietser28, 2023-06-04:
Works. Thank you!

closed 2023-06-04 by fietser28