eez-open/studio #270: EEZ Flow: make string formatting function String.format(...) similar to sprintf.

fietser28, 2023-06-15:
Nice to have and also see discussion in #196

fietser28, 2024-06-15:
[d3](https://github.com/d3/d3-format/tree/v1.4.5#d3-format) might be an option for dashboards. Plotly is also using this.
The nice thing about d3 is that is has native support for SI prefixes.

mvladic, 2024-06-15:
You mean this: https://d3js.org/d3-format ?

fietser28, 2024-06-15:
> You mean this: https://d3js.org/d3-format ?

Yes

mvladic, 2024-07-15:
Implemented as described in my previous comment.

fietser28, 2024-07-16:
I did some research on the background of ```d3-format```:

It implements (and extends) the Python [formatspec](https://docs.python.org/3/library/string.html#formatspec). 
In C++ there is a library that also implements ```formatspec``` named [fmt](https://fmt.dev/latest/). This requires C++. I did not find a pure C implementation. Don't know if this is a show stopper.

Important note: 
For engineering applications (like measurement/SCPI instruments) the ```formatPrefix(specifier, value)``` extension of ```d3-format``` is really useful to display values with an SI-prefix. I did not find the equivalent in ```fmt``` or Python's ```formatspec```, but it can probably be added.

mvladic, 2024-07-16:
> I did not find a pure C implementation. Don't know if this is a show stopper.

Not necessary. I only prefer C libraries over C++ as they are often more lightweight (this is important for embedded usage) and simpler to use. 

I will check fmt library and see if it is suitable for our purpose. We can add a #ifdef to disable it and fallback to sprintf in case when it is too much for the target embedded platform.

> For engineering applications (like measurement/SCPI instruments) the formatPrefix(specifier, value) extension of d3-format is really useful to display values with an SI-prefix

We can add this for the dashboard projects only at first.

mvladic, 2024-07-18:
The function [`String.formatPrefix`](https://d3js.org/d3-format#formatPrefix) is added for the Dashboard projects

bentron2000, 2024-09-20:
Have noticed a bug in version 0.16.1 in the implementation of String.format 

To reproduce:
1. Create a label on any screen.
2. Make the label an expression with `String.format(".2f",1.1) `
3. In `EDIT` mode, observe the label correctly display as `1.10`
4. Switch to `RUN` or `DEBUG` mode, observe the label display as `.2f`

B

mvladic, 2024-09-20:
Format specification for the `String.format` function when used in LVGL project must use `%` prefix. So, your expression should be `String.format("%.2f",1.1)`.

closed 2026-02-09 by mvladic

zalexzperez, 2026-03-24:
This function is limited to just one specifier, right? 
I was hoping to be able to log 5 variables at once.