eez-open/studio #271: Set theme action

prasimix, 2023-06-16:
We need `Set theme` action that will enable a theme to be set from the project.

tylercamp, 2024-06-08:
+1, was going to request a "color palette" feature, but this would work too

yaredna, 2024-10-12:
Looking forward to revision 0.19.0, as the lack of theme in EEZ Studio / LVGL is a show stopper for now. Putting migration project on hold.

mvladic, 2024-10-17:
Implemented.

yaredna, 2024-11-01:
Hello mvladic: 

Two questions:
. color themes: still on track for next month ? 
. in LVGL 8.4, we can't retrieve the needle indicator from a meter to set the value. Do you have a work around to expose the indicator as a global variable or something similar ? Best regards !

mvladic, 2024-11-01:
> . color themes: still on track for next month ?

Yes, color themes are already implemented.

> in LVGL 8.4, we can't retrieve the needle indicator from a meter to set the value. Do you have a work around to expose the indicator as a global variable or something similar ? Best regards !

It doesn't make sense to discuss needle indicator in this issue. It is best to repeat this question on [Discord](https://discord.com/channels/511131497801973761/1207347896857530400).

closed 2024-11-04 by mvladic

zalexzperez, 2026-10-16:
I just started using this feature. 
However, my device crashes when selecting a different theme in the dropdown.

This is because EEZ is trying to update all the screens at once that contain the associated colors, but some of those screens may not be created at start and/or they were deleted on unload. 

As a test, I enabled **Create at start** and disabled **Delete on unload** for all my screens containing the colors on my themes and it no longer crashes.

This is a major limitation.