eez-open/studio #272: no response to IDN query

EdizonTN, 2023-06-26:
Hello,
I have a problem with create my own Instrument.

In YAT serial terminal I try to send IDN command to instrument - Result was OK:
[image]

In EEZ Studio into my created instrument I enter string **"SN:V1.80;"** into IDN field (also with quotation marks).
Instrumet was checked and installed as extension.

In new project I try to add this instrument and try to connect - without success, with message "no response to IDN query".
But in command history I see an answer (see screenshot)! 
Pls, Where is a problem?
EEZ Studio: Version 0.9.99, OS: Windows 10.
Connected instrument: Twintex TP-3303U - Multiple Output Programmable Linear DC Power Supply
Connected via USB cable (In Twintex is a FTDI USB/TTL Serial converter), in EEZ selected serial comm.

[image]

mvladic, 2023-06-27:
If this instrument is using <CR> as line end indicator then that explains why the IDN detection does not work - Studio is expecting <LF> as line end indicator.

I built a version which should also work with <CR> as line end indicator, you can download it from here:

https://www.dropbox.com/s/x2d4c8by0hv3jgg/EEZ%20Studio%20Setup%200.10.1.exe?dl=0

EdizonTN, 2023-06-27:
Perfect!
It's work now!
Thanks!

[image]

closed 2023-06-27 by EdizonTN