eez-open/studio #274: Add multilanguage support in LVGL project

prasimix, 2023-07-01:


mverch67, 2024-06-08:
I think basic support multilanguage would be quite easy to achieve as the text label `"label"` just would need to be generated as `_("label") `instead, so that this `_(const char*)` function is called to translate the string. Either a simple checkbox to indicate translation or using the already available dropdown menu next to the text label could be used.

The strings are handled offline in .yml files and a node script is used to generate the C code. The include of "lv_i18n.h" can be manually added into screens.h.

Reference to lv_i18n:
https://github.com/lvgl/lv_i18n

Reference to squareline studio's usage:
https://docs.squareline.io/docs/multilanguage/

mvladic, 2024-06-08:
We also need to add support through expressions. So you can write something like this:

```
T"Voltage: " + volt_var
```

Here syntax will be `T"..."` since this is already how we use it in EEZ-GUI and Dashboard projects.

mvladic, 2024-06-13:
We should support LVGL YML format for lv_i18n.h:

- Import from YML into Studio so translations can be edited inside Studio
- Export to YML
- Synchronization with the YML file
- Generate source code from YML file inside studio, so there is no need to execute [lv_i18n script](https://github.com/lvgl/lv_i18n) manually

mverch67, 2024-06-16:
I'd still like to propose to do this in two steps in case this project is low on developers:
Step 1- add checkbox (or extend the dropdown) next to the label text and generate _("label") for lvgl C code
Step 2- add _p("label", cnt) to UI and code generation as well as import/export YML files and YML code-generation

If the tiny little effort for step 1 is done right away it has the big advantage that migrants from squareline studio can use this feature right away as they did in the past without losing this functionality for their project. Step 1 is needed anyway and may not change much when developing step 2 at a later stage.

mvladic, 2024-06-16:
I implemented Step 1 in the master.

In "Label", "Textarea", "Dropdown" and "Roller" widgets you can now select for "Text" and "Options" a new option called "Translated literal":

[image]
[image]

Here is the project I used for testing:

https://github.com/eez-open/lvgl_test_translate_literal

Here is the place where I putted i18n files:

https://github.com/eez-open/lvgl_test_translate_literal/tree/master/src/i18n

I added following line to `screens.c` file template:

[image]

And added these lines in [`main.c`](https://github.com/eez-open/lvgl_test_translate_literal/blob/master/main.c#L72)

dhugoexe, 2025-07-14:
> We also need to add support through expressions. So you can write something like this:
> 
> ```
> T"Voltage: " + volt_var
> ```
> 
> Here syntax will be `T"..."` since this is already how we use it in EEZ-GUI and Dashboard projects.

Where does the development of this feature stand?