eez-open/studio #277: Add options to improve memory usage of LVGL projects

fietser28, 2023-07-30:
Current implementation creates all LVGL screens, widgets and corresponding flow states and variables at startup and keeps them active in memory. This requires a large LVGL memory pool if the amount of screens/widgets grows, even if the screen is only showed once or rarely. Even for simple screens the LVGL memory is considerable.

The same applies to the flow states, but I'm not sure if there is much to be gained here in effort vs. memory usage reduction.

- [ ]  I would like to have an option to delete LVGL screens resources when they are not active. A screen will have a checkbox to ' optimise LVGL memory usage'. E.g. have studio create delete_screen_* functions and make sure  the needed delete/create_screen_* functions are called when switching screens instead of just calling all the create_screen_* functions at startup. This is for just the LVGL objects not the flowstate. I think the implication is that if a screen with this property is not active properties can not be changed (from the LVGL action) and when switching to the screen all the initial LVGL settings are loaded. the flow state (e.g. local variables) are still retained. It is up to the user to set the LVGL states upon re-entry.
- [ ] Also implement the dynamic creation and deletion of corresponding flow states. This will make local variables non-persistent when a screen is deactivated.
- [ ] Nice to have:  Make the Flow QUEUE_SIZE (queue.cpp) user configurable. Allowing for more memory tuning. 

Example use case:
Raspberry Pi Pico with 264KBytes of RAM and 2MBytes of FLASH. The flow has 7 screens (of 320x240) with around 100 widgets, 50 native variables and 25 local variables in total. Static memory allocations are now: 
- the LVGL memory pool (LV_MEM_SIZE + overhead) is already around 100Kbytes and I need to keep increasing it when adding screens/widgets. 
- The eez::flow::g_queue is 11.7KB.
- the LVGL display buffer is set to 6300 bytes. This is working fine and a constant size. This is outside of the flow code and can be tweaked already.

I suspect this is a common issue with a lot of MCU's where RAM is limited.

Point of attention: Will the LVGL memory pool implementation keep the fragmentation low to keep functioning properly with a limited pool size?

mvladic, 2024-06-02:
Flow execution queue size can be adjusted also. By default, memory for 1000 entries is reserved. You can change this to some smaller value by putting the following line in `eez-framework-conf.h` that you can place next to `lv_conf.h`:

```
#define EEZ_FLOW_QUEUE_SIZE 250
```

But, you must determine value that works for you empirically. You can get the max. value used so far by calling `eez::flow::getMaxQueueSize()`.

julian-mjp, 2024-09-09:
I found myself in the same need of delete LVGL screens resources when they are not active, I'm working in a project with a lot of different screens and it would be very helpful.

Do you know any alternative that I could use now?

zalexzperez, 2025-01-19:
I'm also looking forward to the screen management feature.
After creating 29 screens at 320x172 resolution, an LVGL memory pool of 140KB is not enough.

The "optimize LVGL memory usage" checkbox makes sense but I think it would be more powerful if we could create groups of screens, similar to what we already have with encoder groups.
That way, the developer would only have to worry about loading the desired group of screens, instead of manually selecting them.

For example, this would help greatly when you have at least two distinct working modes in your device, each of which with a different set of screens.

mvladic, 2025-01-19:
I created a [separate issue #706](https://github.com/eez-open/studio/issues/706) for this. You can paste your comment there.