eez-open/studio #282: A few possible enhancements found in flow

FractalEngineer, 2023-09-02:
I've been keeping track of possible bugs/enhancements while going through the process of making my own test procedures;

In this thread, I'd like to leave a continued list of enhancements I would have found useful :)

1- Enabling copy of themes & styles with their attributes between different projects (currently pasting themes from a different project defaults all to black color, 
[image]

2- Enabling automatic -or manual with scroll wheel- scrolling of the structure list when dragging an element to the top/bottom of the window (couldn't capture the no go icon)
[image]

3- I know there's the docs, but a contextual help popup to explain widget purposes and uses, along with specific properties (tooltips?) could be a neat quality of life improvement 
[image]

mvladic, 2023-09-02:
> 2- Enabling automatic -or manual with scroll wheel- scrolling of the structure list when dragging an element to the top/bottom of the window (couldn't capture the no go icon)

This is already supported (if I understood correctly):

[image]

FractalEngineer, 2023-09-02:
> 

Ah true, looks like I must have hit the perfectly wrong delay/position a few times in a row to make me believe it wasn't 😅