eez-open/studio #287: Flow: Show/search SCPI documentation while editing SCPI component

fietser28, 2023-09-11:
In the instrument there is a nice SCPI documentation panel that dynamically searches the SCPI documentation if you type.

I would like to have the same panel present (by default at the bottom as an addition tab next to check/output/search/reference) that will search thru the documentation of the instrument I am editing the SCPI for.

It is slightly more complex than the script implementation:
- it needs to remove the <var> = before the scpi command and remove any variable substitions ({var}) in the present line
- determine which documentation to use. There are several options for this: 
  - query the instrument connected and see if an IEXT is present
  - let the user choose in the instrument properties which IEXT to use?
  - combination of both?

mvladic, 2023-09-19:
I have this idea to put command browser here:

[image]

[image]

This is work in progress.

mvladic, 2023-09-22:
This is implemented in the latest version.

fietser28, 2023-10-28:
It is working nice. I noticed a few things when using it. These are minor:

- [ ]  When I start editing an existing SCPI command (and the link button is checked) AND there is 1 SCPI command found the contents/help of this command is not shown. You have to click on it to show the manual info.

- [ ]  When multiple are found I would like to browse (using cursor key down/up?) to the right one instead of using the mouse to select one. Maybe I just don't know the right shortcut.

- [ ]  If the SCPI command (the command string itself not the arguments of the command) is composed using a variable the search will fail:

[image]

Removing the {} part results in:

[image]

During the filtering I assume the part of the command with {} should probably be removed from the search as wel.

mvladic, 2024-04-04:
> When I start editing an existing SCPI command (and the link button is checked) AND there is 1 SCPI command found the contents/help of this command is not shown. You have to click on it to show the manual info.

Can't reproduce this problem.

> If the SCPI command (the command string itself not the arguments of the command) is composed using a variable the search will fail:

This should be fixed now.

mvladic, 2024-04-05:
> When multiple are found I would like to browse (using cursor key down/up?) to the right one instead of using the mouse to select one. Maybe I just don't know the right shortcut.

Implemented. F3 and Shift+F3 are keyboard shortcuts.

fietser28, 2024-04-06:
Thank you. I also can't reproduce the first item.

closed 2024-04-08 by prasimix