eez-open/studio #298: A JavaScript error occured in the main process

demondev, 2023-11-17:
Hi,

I have just downloaded and installed this https://github.com/eez-open/studio/releases/download/v0.10.4/EEZ.Studio.Setup.0.10.4.exe and followed the Windows install instructions here https://github.com/eez-open/studio#windows

When lauching I am confronted with the following error.

[image]

Any ideas ?

demondev, 2023-11-17:
Removing 0.10.4 then installing 0.10.3 and the application loads fine.

mvladic, 2023-11-18:
0.10.4, while still in development, was accidentally released. Unfortunately, it was also broken build. Until next official build, we recommend to use 0.10.3.

demondev, 2023-11-18:
Understood, thanks for the response.

closed 2023-11-18 by demondev