eez-open/studio #299: BB3-Dashboard Slot2 (3rdslot) fails to set U or I. Numberinput doesn't disappear. Channel 1 and 2 work fine

ronnyvs1, 2024-01-02:
Hi all,
Already thanks for helping... and an amazing tool!!!!
I did make a change from the BB3-dashboard, to make it work with Rigol power supply's.
actually everything seems to work fine...
Channel 1 en 2 are working great... ,

Channel 3 output enable also works,  meters showing right values etc.. but....
When i want to set channel 3 it has the problem that it won't set the voltage or current.
when i hit the "Ok"-button, the settings are not being managed by EEZ-studio. The numberinput should disappear but stays visible.
also the number input "is_open" stays "true"
It also gives the next error message:

**Error: Failed to evaluate variable no. 1 in SetVariable: Array value expected**

 (see image)

in the background everything else stil works, but i can't control it , because the numberinput stays visible...

Hope someone can help me..
friendly regards,
Ronny
[image]

mvladic, 2024-01-02:
Congratulations, you found a bug! 

In `NumberInput` user widget, `SetVariable` should be corrected (add `- 1` in two places):

[image]

I will fix it (officially) next week when I'm back from vacation.

ronnyvs1, 2024-01-02:
Thank you for your help!
Now it works very well...
i made it , to post it, so that other people can remote control the latest Rigol power supply's.
I only want to do the basics, so that other people can modify it with more options etc.

maybe a last question,
How can i convert this from a project to an instrument and i want to add the Connect-Disconnect button...?

but for now, enjoy your vacation!!!

Friendly regards and thanks for your help!!
Ronny

mvladic, 2024-01-11:
> How can i convert this from a project to an instrument and i want to add the Connect-Disconnect button...?

There is an option for the instrument project (IEXT) to include one or more dashboard projects. For example, here is how BB3 dashboard is added to the BB3 IEXT:

[image]

When dashboard is added to the instrument there is no need to add separate Connect/Disconnect buttons as regular instrument Connect/Disconnect buttons will be used:

[image]

mvladic, 2024-04-16:
Fixed.

closed 2024-05-06 by mvladic