eez-open/studio #300: Flex flow is missing

dolence, 2024-02-06:
I just have this software installed and my impressions are so good! I was skeptical at first, but this software everything SquareLine Studio should be.  It's been a couple of minutes of pure fun testing it. Everything works as expected and the workflow is a breeze. 
One thing I'm missing is the concept of Flex Flow present on the tool aforementioned. Are there any plans to have this implemented?
Also, latest version is crashing on Ubuntu 22.03. I had to download  and install the previous one.

prasimix, 2024-02-07:
Thank you for your interest. We are continuously working on improving EEZ Studio and we will take your request into account, especially if there are others interested in the same feature. Thanks in advance for further evaluation and feel free to add new tasks and comments.

dolence, 2024-02-09:
That's nice! Let me expand a little further. Flex would be good for everyone using various display with different resolutions or content being generated dynamically. 
Flex grows when resized, auto placing the items in such a way they fill the remaining space inside it, much like Flexbox does to responsive sites.

[Flex in LVGL documentation](https://docs.lvgl.io/master/layouts/flex.html)

dolence, 2024-02-20:
Some more information regarding this:

How it is:
[image]

How it is/was on SquareLine:
[image]

Figma mockup:
[image]

mvladic, 2024-03-08:
Support for the flex layout is now implemented in the master branch.

Since all the Flex-related values are style properties under the hood, this feature is configured in the Styles section where a new group called LAYOUT is added:

[image]

dolence, 2024-03-09:
This is nice. I will compile it fro msource, test and report back. Thank you very much!