eez-open/studio #301: Add lv_tabview

dolence, 2024-02-07:
The object `lv_tabview` used to organize the content using tabs is missing.

Example: [lv_tabview](https://docs.lvgl.io/master/widgets/tabview.html)

Jvsjvs, 2024-06-27:
Please also add the: "Container" object. This is used frequently to create an hierarchy in the UI layout structure (e.g. in the tabview).

Thanks

mvladic, 2024-07-10:
> Please also add the: "Container" object. This is used frequently to create an hierarchy in the UI layout structure (e.g. in the tabview).

In the latest version I added Container widget exactly for the purpose you described. So you can add two container widgets to the TabView, one for the Bar and another for the Content. When you do this then you should put Tab widgets under the Content. This configuration is useful when you want to style Bar and Content part of the TabView. This is optional, if you don't need to style TabView then you can place Tab widgets immediately under TabView.

[image]

Jvsjvs, 2024-07-11:
Great, thanks!

closed 2024-08-05 by prasimix