eez-open/studio #302: Flow: 'Find All references' displays actions not actually connected

fietser28, 2024-02-10:
Symptom: Searching for all references to a specific action in the project results in a lot of incorrect hits. It is an inconvenience in the UI, the flows seem to work fine.

Example with a single error (the search was for find all references to dummyAction):

[image]

Attached is the project. It was created as a new project with latest GitHub version. 
[referencesbug.eez-project.zip](https://github.com/eez-open/studio/files/14228785/referencesbug.eez-project.zip)

Looking in the eez-project file, it seems all evenHandlers of handlerType Flow also have a filled action field. This looks incorrect to me:
Event handler section of the 'Hello, world!' label:
```
"eventHandlers": [
            {
              "objID": "ad28c923-1aac-422f-d9d0-1a48cc02fed7",
              "eventName": "PRESSED",
              "handlerType": "flow",
              "action": "dummyAction"
            }
          ],
```

In larger projects the action fields are all filled with a reference to the same action.

mvladic, 2024-02-12:
Fixed.

fietser28, 2024-02-12:
Works, thank you!

closed 2024-02-12 by fietser28