eez-open/studio #311: Add support for LVGL 9.0

mvladic, 2024-02-16:
For now, add support for all currently implemented widgets and styles to work with LVGL 9.0. It should be possible to choose between versions 8.3 and 9.0 for new and existing projects.

mvladic, 2024-03-09:
**Important!**

[Online Image Converter](https://lvgl.io/tools/imageconverter) and correspondingly [lv_img_conv library](https://github.com/lvgl/lv_img_conv) do not support LVGL 9.0. Instead, [LVGLImage.py](https://github.com/lvgl/lvgl/blob/master/scripts/LVGLImage.py) python script is used. 

LVGLImage.py script is distributed with the EEZ Studio, but for EEZ Studio to be able to execute this script following requirements must be fulfilled:

1. Python 3 is installed on the system

2. Install `pypng` and `lz4` modules:

      ```
      python3 -m pip install pypng
      python3 -m pip install lz4
      ```

3. Install `pngquant` tool:

      -   Linux (Debian)
      
          ```
          sudo apt install pngquant
          ```
      
      -   Mac
      
          ```
          brew install pngquant
          ```
      
      -   Windows
      
          No action needed, it is distributed with the EEZ Studio.

closed 2024-03-11 by prasimix

mhaberler, 2024-04-05:
Hi,
moving from 8 to 9 as well, and migrated from [PubSub to Observers ](https://github.com/lvgl/lvgl/issues/5419) which work welll

do Observers have any bearing on Studio 8->9 migration?

mvladic, 2024-04-05:
No, because EEZ Studio and eez-framework don't use PubSub or Observers.

liriyao, 2024-07-24:
[image]

itarozzi, 2024-07-24:
> i have already install pypng and lz4 ,but it still can't build when i have a PNG in the srceen 

In this case seems pypng is well installed, but it encounter some errors in png format. Can you share the png image you are using?

liriyao, 2024-07-24:
> > i have already install pypng and lz4 ,but it still can't build when i have a PNG in the srceen
> 
> In this case seems pypng is well installed, but it encounter some errors in png format. Can you share the png image you are using?

i just use the wechat srceen shot to make a png like this 
[image]

liriyao, 2024-07-24:
> > i have already install pypng and lz4 ,but it still can't build when i have a PNG in the srceen
> 
> In this case seems pypng is well installed, but it encounter some errors in png format. Can you share the png image you are using?

I used another screenshot method windows  screenshot is OK,wechat screenshot don't .Is it a compatibility issue?

liriyao, 2024-07-24:
[image]
fuck the wechat !