eez-open/studio #312: Copying files UI integrating files/file paths and placement

Gert69, 2024-02-16:
I recently moved from Square Line Studio to EEZ Studio everything is the same but different.

I use the waveshare ESP32 S3 Development board https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3

Since Square Line Studio doesn't support my board I use:
ESP32_Display_Panel
ESP32_IO_Expander
lvgl
ESP_Panel_Conf
lv_conf

I use the conf files to edit the display drivers touch driver ect.

Attached is a Folder where I normally paste my UI files that square Line Studio creates into and it loads up to my development board and displays.
[image]

Can someone please point me in the rite direction in where to add EZZ UI files as i am using arduino to upload to my development board.

Thank you in advance

goran-mahovlic, 2024-02-16:
Hi, on new version EEZ Studio has this option - and that could be what you are looking for.

https://github.com/eez-open/studio/releases/tag/untagged-c78c3b513df2c4770a4c

So download latest build, and then New project and select LVGL - you can select location where UI will be after you pres build inside studio files should be in your working directory...

[image]

goran-mahovlic, 2024-02-16:
And feel free to join eez-open discord channel: https://discord.gg/axzGY7Wvnq

Gert69, 2024-02-16:
Hi @goran-mahovlic  I have managed to create a demo through that example you have provide.

[image]

So I copied the UI files across like with Square Line but obviously I am get Compilation errors as i know the files are not placed in the correct matter and folders I'm pretty much  screwed and have no idea where the place what to get it going again pretty new to this and learned the hard way how to get Square Line going and used to SL pretty dumb now with EZZ

mvladic, 2024-02-16:
> I am get Compilation errors as i know the files are not placed in the correct matter.

Post here those errors you get when compiling.

Gert69, 2024-02-16:
My Code:

`#include <Arduino.h>
#include <ESP_Panel_Library.h>
#include <ESP_IOExpander_Library.h>
#include <ui.h>

/* LVGL porting configurations */
#define LVGL_TICK_PERIOD_MS     (2)
#define LVGL_TASK_MAX_DELAY_MS  (500)
#define LVGL_TASK_MIN_DELAY_MS  (1)
#define LVGL_TASK_STACK_SIZE    (4 * 1010)
#define LVGL_TASK_PRIORITY      (2)
#define LVGL_BUF_SIZE           (ESP_PANEL_LCD_H_RES * 8)

ESP_Panel *panel = NULL;
SemaphoreHandle_t lvgl_mux = NULL;                  // LVGL mutex

#if ESP_PANEL_LCD_BUS_TYPE == ESP_PANEL_BUS_TYPE_RGB
/* Display flushing */
void lvgl_port_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
    panel->getLcd()->drawBitmap(area->x1, area->y1, area->x2 + 1, area->y2 + 1, color_p);
    lv_disp_flush_ready(disp);
}
#else
/* Display flushing */
void lvgl_port_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
    panel->getLcd()->drawBitmap(area->x1, area->y1, area->x2 + 1, area->y2 + 1, color_p);
}

bool notify_lvgl_flush_ready(void *user_ctx)
{
    lv_disp_drv_t *disp_driver = (lv_disp_drv_t *)user_ctx;
    lv_disp_flush_ready(disp_driver);
    return false;
}
#endif /* ESP_PANEL_LCD_BUS_TYPE */

#if ESP_PANEL_USE_LCD_TOUCH
/* Read the touchpad */
void lvgl_port_tp_read(lv_indev_drv_t * indev, lv_indev_data_t * data)
{
    panel->getLcdTouch()->readData();

    bool touched = panel->getLcdTouch()->getTouchState();
    if(!touched) {
        data->state = LV_INDEV_STATE_REL;
    } else {
        TouchPoint point = panel->getLcdTouch()->getPoint();

        data->state = LV_INDEV_STATE_PR;
        /*Set the coordinates*/
        data->point.x = point.x;
        data->point.y = point.y;

        // Serial.printf("Touch point: x %d, y %d\n", point.x, point.y);
    }
}
#endif

void lvgl_port_lock(int timeout_ms)
{
    const TickType_t timeout_ticks = (timeout_ms < 0) ? portMAX_DELAY : pdMS_TO_TICKS(timeout_ms);
    xSemaphoreTakeRecursive(lvgl_mux, timeout_ticks);
}

void lvgl_port_unlock(void)
{
    xSemaphoreGiveRecursive(lvgl_mux);
}

void lvgl_port_task(void *arg)
{
    Serial.println("Starting LVGL task");

    uint32_t task_delay_ms = LVGL_TASK_MAX_DELAY_MS;
    while (1) {
        // Lock the mutex due to the LVGL APIs are not thread-safe
        lvgl_port_lock(-1);
        task_delay_ms = lv_timer_handler();
        // Release the mutex
        lvgl_port_unlock();
        if (task_delay_ms > LVGL_TASK_MAX_DELAY_MS) {
            task_delay_ms = LVGL_TASK_MAX_DELAY_MS;
        } else if (task_delay_ms < LVGL_TASK_MIN_DELAY_MS) {
            task_delay_ms = LVGL_TASK_MIN_DELAY_MS;
        }
        vTaskDelay(pdMS_TO_TICKS(task_delay_ms));
    }
}

void setup()
{
    Serial.begin(115200); /* prepare for possible serial debug */

    String LVGL_Arduino = "Hello Squareline!";
    LVGL_Arduino += String('V') + lv_version_major() + "." + lv_version_minor() + "." + lv_version_patch();

    Serial.println(LVGL_Arduino);
    Serial.println("I am ESP32_Display_Panel");

    panel = new ESP_Panel();

    /* Initialize LVGL core */
    lv_init();

    /* Initialize LVGL buffers */
    static lv_disp_draw_buf_t draw_buf;
    /* Using double buffers is more faster than single buffer */
    /* Using internal SRAM is more fast than PSRAM (Note: Memory allocated using `malloc` may be located in PSRAM.) */
    uint8_t *buf = (uint8_t *)heap_caps_calloc(1, LVGL_BUF_SIZE * sizeof(lv_color_t), MALLOC_CAP_INTERNAL);
    assert(buf);
    lv_disp_draw_buf_init(&draw_buf, buf, NULL, LVGL_BUF_SIZE);

    /* Initialize the display device */
    static lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    /* Change the following line to your display resolution */
    disp_drv.hor_res = ESP_PANEL_LCD_H_RES;
    disp_drv.ver_res = ESP_PANEL_LCD_V_RES;
    disp_drv.flush_cb = lvgl_port_disp_flush;
    disp_drv.draw_buf = &draw_buf;
    lv_disp_drv_register(&disp_drv);

#if ESP_PANEL_USE_LCD_TOUCH
    /* Initialize the input device */
    static lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = lvgl_port_tp_read;
    lv_indev_drv_register(&indev_drv);
#endif

#if defined(ESP_PANEL_BOARD_ESP32_S3_LCD_EV_BOARD) || defined(ESP_PANEL_BOARD_ESP32_S3_KORVO_2)
    /**
     * These development boards require the use of an IO expander to configure the screen,
     * so it needs to be initialized in advance and registered with the panel for use.
     *
     */
    Serial.println("Initialize IO expander");
    /* Initialize IO expander */
    ESP_IOExpander *expander = new ESP_IOExpander_TCA95xx_8bit(ESP_PANEL_LCD_TOUCH_BUS_HOST_ID, ESP_IO_EXPANDER_I2C_TCA9554_ADDRESS_000, ESP_PANEL_LCD_TOUCH_I2C_IO_SCL, ESP_PANEL_LCD_TOUCH_I2C_IO_SDA);
    expander->init();
    expander->begin();
    /* Add into panel */
    panel->addIOExpander(expander);
#endif

    /* Initialize bus and device of panel */
    panel->init();
#if ESP_PANEL_LCD_BUS_TYPE != ESP_PANEL_BUS_TYPE_RGB
    /* Register a function to notify LVGL when the panel is ready to flush */
    /* This is useful for refreshing the screen using DMA transfers */
    panel->getLcd()->setCallback(notify_lvgl_flush_ready, &disp_drv);
#endif
    /* Start panel */
    panel->begin();

    /* Create a task to run the LVGL task periodically */
    lvgl_mux = xSemaphoreCreateRecursiveMutex();
    xTaskCreate(lvgl_port_task, "lvgl", LVGL_TASK_STACK_SIZE, NULL, LVGL_TASK_PRIORITY, NULL);

    /* Lock the mutex due to the LVGL APIs are not thread-safe */
    lvgl_port_lock(-1);

    ui_init();

    lvgl_port_unlock();

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("Loop");
    sleep(1);
}
`

Gert69, 2024-02-16:
My Error:

`c:/users/esp/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\esp\AppData\Local\Temp\arduino\sketches\874C0DC56262767F71AC58FBED408312\sketch\objs.a(Porting.ino.cpp.o):(.literal._Z5setupv+0x74): undefined reference to `ui_init()'
c:/users/esp/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\esp\AppData\Local\Temp\arduino\sketches\874C0DC56262767F71AC58FBED408312\sketch\objs.a(Porting.ino.cpp.o): in function `setup()':
C:\Users\esp\Desktop\Square\libraries\ESP32_Display_Panel\examples\SquareLine\Porting/Porting.ino:166: undefined reference to `ui_init()'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1`

mvladic, 2024-02-16:
This is linker error which means you didn't compile `ui.c` and probably all other files `.c` that are generated by the EEZStudio.

mvladic, 2024-02-16:
Maybe you should place generated source files (ui.c, etc) in the same folder where Porting.ino is.

Gert69, 2024-02-16:
@mvladic got the other issue sorted out I hope!

Now I'm getting the next error message.

`c:\Users\esp\Desktop\Square\libraries\ui\ui.c:4:10: fatal error: flow_def.h: No such file or directory
 #include "flow_def.h"
          ^~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1`

Gert69, 2024-02-16:
If I remove flow def I get the first compilation error I have posted

goran-mahovlic, 2024-02-16:
Did you use LVGL template? not LVGL FLOW

goran-mahovlic, 2024-02-16:
also it may not overwrite some files, so best is if you now delete previously  created UI folder and recreate it with only LVGL template

mvladic, 2024-02-16:
> Now I'm getting the next error message.
> `c:\Users\esp\Desktop\Square\libraries\ui\ui.c:4:10: fatal error: flow_def.h: No such file or directory
> #include "flow_def.h"
> ^~~~~~~~~~~~
> compilation terminated.

This is bug in EEZ Studio! It shouldn't generate `#include "flow_def.h"`. You can remove it in your project:

[image]

Gert69, 2024-02-16:
@mvladic  thanks martin removed it as advised.

This is the result after removing it.

`c:/users/esp/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\esp\AppData\Local\Temp\arduino\sketches\874C0DC56262767F71AC58FBED408312\sketch\objs.a(Porting.ino.cpp.o):(.literal._Z5setupv+0x74): undefined reference to `ui_init()'
c:/users/esp/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\esp\AppData\Local\Temp\arduino\sketches\874C0DC56262767F71AC58FBED408312\sketch\objs.a(Porting.ino.cpp.o): in function `setup()':
C:\Users\esp\Desktop\Square\libraries\ESP32_Display_Panel\examples\SquareLine\Porting/Porting.ino:196: undefined reference to `ui_init()'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1`

mvladic, 2024-02-16:
Yes, this is the same linking error. I tells me that `ui.c` is not compiled along with `Porting.ino`. You should place `ui.c` in the same directory where `Porting.ino` is.

mvladic, 2024-02-16:
To be clear, not only `ui.c`, but all the files generated by the Studio should be placed in the same directory where `Porting.ino` is. This is required because of the peculiarities of the Arduino building system.

mvladic, 2024-02-16:
You can place `.eez-project` file also in the same folder where `Porting.ino` is and then select for build destination folder `.` so it will also generate sources in that same folder so you don't need to move files manually afer each build in Studio:

[image]

Gert69, 2024-02-16:
@mvladic Thank you I am grateful for the advice and help truly appreciated I will give this a go and respond to you @mvladic.

Apologies this is all new to me and I am learning as I go.

closed 2024-04-15 by mvladic

drik, 2024-07-19:
eez-studio is very interesting tool. I think it deserve moore visibility, more attention from makers community, espacially those who are interested in LVGL. It make my project speed up 100000%. @mvladic, your comments are very useful. It make me achieve integration with my project and now I can take my project to a new level. Big thanks to you and all the contributors, keep doing the great job...