eez-open/studio #314: LVGL Flow: Add native API for changing screens

fietser28, 2024-02-17:
In native code I would like to have an API to change to a specific screen or return to the previous screen (upon next ui_tick() )
It should be clear if this API is thread safe with respect to the ui_tick() call.

Nice to have: be able to query the current screen.

mvladic, 2024-02-17:
Following API functions added to the `<eez/core/lvgl_api.h>`:

```
void eez_flow_set_screen(int16_t screenId, lv_scr_load_anim_t animType, uint32_t speed, uint32_t delay);
void eez_flow_push_screen(int16_t screenId, lv_scr_load_anim_t animType, uint32_t speed, uint32_t delay);
void eez_flow_pop_screen(lv_scr_load_anim_t animType, uint32_t speed, uint32_t delay);
```

Studio will generate enum ScreensEnum { ... }, for example:

```
enum ScreensEnum {
    SCREEN_ID_SCREEN_1_PRINT = 1,
    SCREEN_ID_SCREEN_2_MOVE = 2,
    SCREEN_ID_SCREEN_3_SETTING = 3,
};
```

- `eez_flow_set_screen`: sets active screen (pass value from `ScreensEnum` as `screenId`). This function clears screens stack.
- `eez_flow_push_screen`: pushes currently active screen id on the screens stack and sets active screen to the given screenId
- `eez_flow_push_screen`: sets active screen to the screenId from the top of the stack

mvladic, 2024-02-17:
Done.

fietser28, 2024-02-18:
Can confirm it works on MCU.

closed 2024-02-19 by prasimix