eez-open/studio #316: LVGL Flow: Add event handling for pages/screens

fietser28, 2024-02-17:
LVGL has several events specific to screens (load/unload). These events can currently not be used in flow because there is no event handling and interface for it.

It is convenient to have this interface available to trigger actions/flows when a screen is loaded (displayed) or when left.

In the UI this probably means:
- the properties tab of a screen get an events section similar to other widgets.
- If flow events are defined in the properties handles appear at the screen borders, similar to other widgets.

mvladic, 2024-05-29:
Done.

fietser28, 2024-06-05:
2 observations:
- It is possible to add screen_(un)load(ed) events to widgets (like buttons/panels). In my interpretation of the LVGL documentation this is fine but probably not intentional. It becomes a problem:
- When I create a screen flow event in a button and later a screen event for the screen then making the connection in the editor connects to the wrong handle:

[image]
In this example I pressed on the handle of the screen but the line to the button appears (the mouse crosshair is not in the screenshot but at the floating line). The button handle is the event shown in the properties.

Removing the event on the button allows me to connect to the screen.

I don't see a real use case for connecting handles to widgets, so this probably should be disabled. 
(I made a mistake in a different project and had to find the culprit in the eel-project file manually)

Apart from this, it works in the simulator. Will also test this on real hardware.

mvladic, 2024-06-05:
Fixed.

This bug is not specific to screen_loaded event or any other screen event. It can happen with any event. I'm surprised it wasn't discovered before.

fietser28, 2024-06-05:
Everything is working now! Also tested on hardware.

Does this feature need documentation? Otherwise it can be closed.

closed 2024-06-06 by mvladic