eez-open/studio #318: LVGL: Allow UTF8 code for custom fonts on Text field

dolence, 2024-02-20:
I don't know if it is something already implemented but I have a custom font with symbols from Google Fonts/Material Design and I don't know how to reference to individual characters on my label Text field. LVGL handle this using the hex code converted to UTF-8.

[image]

[image]

[image]

fietser28, 2024-02-20:
Please try: \ue5d2

mvladic, 2024-02-20:
Also, I'm not sure it will work with literal. If it doesn't work change to expression, i.e.: `"\ue5d2"`

dolence, 2024-02-21:
> Also, I'm not sure it will work with literal. If it doesn't work change to expression, i.e.: `"\ue5d2"`

This worked! Thank you!

mvladic, 2024-04-08:
We should support unicode escape (i.e. \ue5d2) also for the literal strings.

mvladic, 2024-05-06:
Done. Unicode escape (i.e. \ue5d2) is now supported also for the literal strings.

closed 2024-05-06 by mvladic

claudobahn, 2025-04-11:
@mvladic support for unicode escaping appears to have broken w/ 0.23.0 release, specifically with the following commit, I believe: https://github.com/eez-open/studio/commit/c0b2d433393901a2b5259a8970e9d97b881995c6#diff-2818504636f9b20a4b553303c54b052ba7a7d916e47c903132e539ea0e1f919cR100

The chain of `else if` statements probably needs a `else if (ch == "\u")` in addition to `\n`, `\r`, `\t`.

mvladic, 2025-04-13:
Another user reported the same problem, see #783.