eez-open/studio #320: ACTIONS not showing when setting event

Syrinx98, 2024-02-22:
This is the screenshot of what i see if i try to add an event to a widget, i also tried with other widgets, same problem, the dropdown list seems not to display anything

[image]

mvladic, 2024-02-22:
Ok, I found what is problem. This Action dropdown is empty because you didn't define any User Action, here:

[image]

Syrinx98, 2024-02-22:
Oh ok, but what is the user action? a function in c the code?

mvladic, 2024-02-22:
Yes. You created project with EEZ Flow, so your only option for event handler is C function.

Syrinx98, 2024-02-22:
Perfect, thank you for your speed and your support, and sorry for the silly question (i'm trying to pass from squareline to eez studio, since the recent news).

Syrinx98, 2024-02-22:
is there any way to change screen with eez studio? without using eez flow?

mvladic, 2024-02-22:
For example:

```
#include ".../ui/screens.h"

lv_scr_load_anim(objects.main, LV_SCR_LOAD_ANIM_FADE_IN, 200, 0, false);
```
Every page you add will be in objects variable (check `screens.h` header file). Also, every widget with Name property set will be also added to objects variable:

[image]

Syrinx98, 2024-02-22:
Yes, this i know, what i'm saying is: without eez flow there is no possibility to see a screen change in  EEZ studio, right? if i have to try the interface i cannot do that in the program, right?

mvladic, 2024-02-22:
Yes, you are right.

Syrinx98, 2024-02-22:
Thank you however, you are very kind!

mvladic, 2024-02-22:
We concentrated on EEZ Flow as far as Simulator capabilities are concerned. There, interaction with the UI is possible while you are inside Flow. But I think something will need to be done for this scenario when Flow is not used, so that if nothing else, it is possible to switch from screen to screen.

closed 2024-04-15 by mvladic