eez-open/studio #326: Expression Builder blank on Mac (Intel) Build

janbernrader, 2024-03-06:
The expression builder fails to display on mac in the current version.
In the developer console the following error is shown:

Uncaught TypeError: Cannot read properties of undefined (reading 'getEditor')
    at _a.render (/private/var/folders/zk/5n70z3j90tjgckpqv_mzyxfm0000gn/T/AppTranslocation/CAB051D6-9824-42F7-A586-B8875D8694A2/d/EEZ Studio.app/Contents/Resources/app.asar/build/project-editor/flow/expression/ExpressionBuilder.js:1:19214)
    at Object.Q [as _allowStateChanges] (/private/var/folders/zk/5n70z3j90tjgckpqv_mzyxfm0000gn/T/AppTranslocation/CAB051D6-9824-42F7-A586-B8875D8694A2/d/EEZ Studio.app/Contents/Resources/app.asar/node_modules/mobx/dist/mobx.cjs.production.min.js:1:7227)
    at /private/var/folders/zk/5n70z3j90tjgckpqv_mzyxfm0000gn/T/AppTranslocation/CAB051D6-9824-42F7-A586-B8875D8694A2/d/EEZ Studio.app/Contents/Resources/app.asar/node_modules/mobx-react/dist/mobxreact.cjs.production.min.js:1:4284
    at rt (/private/var/folders/zk/5n70z3j90tjgckpqv_mzyxfm0000gn/T/AppTranslocation/CAB051D6-9824-42F7-A586-B8875D8694A2/d/EEZ Studio.app/Contents/Resources/app.asar/node_modules/mobx/dist/mobx.cjs.production.min.js:1:7921)
    at n.track (/private/var/folders/zk/5n70z3j90tjgckpqv_mzyxfm0000gn/T/AppTranslocation/CAB051D6-9824-42F7-A586-B8875D8694A2/d/EEZ Studio.app/Contents/Resources/app.asar/node_modules/mobx/dist/mobx.cjs.production.min.js:1:25857)
    at _a.e [as render] (/private/var/folders/zk/5n70z3j90tjgckpqv_mzyxfm0000gn/T/AppTranslocation/CAB051D6-9824-42F7-A586-B8875D8694A2/d/EEZ Studio.app/Contents/Resources/app.asar/node_modules/mobx-react/dist/mobxreact.cjs.production.min.js:1:4258)
    at r.render (/private/var/folders/zk/5n70z3j90tjgckpqv_mzyxfm0000gn/T/AppTranslocation/CAB051D6-9824-42F7-A586-B8875D8694A2/d/EEZ Studio.app/Contents/Resources/app.asar/node_modules/mobx-react/dist/mobxreact.cjs.production.min.js:1:3197)
    at kj (/private/var/folders/zk/5n70z3j90tjgckpqv_mzyxfm0000gn/T/AppTranslocation/CAB051D6-9824-42F7-A586-B8875D8694A2/d/EEZ Studio.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.production.min.js:204:189)
    at ij (/private/var/folders/zk/5n70z3j90tjgckpqv_mzyxfm0000gn/T/AppTranslocation/CAB051D6-9824-42F7-A586-B8875D8694A2/d/EEZ Studio.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.production.min.js:203:149)
    at Wk (/private/var/folders/zk/5n70z3j90tjgckpqv_mzyxfm0000gn/T/AppTranslocation/CAB051D6-9824-42F7-A586-B8875D8694A2/d/EEZ Studio.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.production.min.js:292:172)

janbernrader, 2024-03-06:
The same work around as in issue #310 works, but would be great if the mentioned fix for windows could also be applied to the other versions. (Still present in macOS 10.6)

Thanks for your awesome work, much appreciated!

mvladic, 2024-03-06:
Hm, it seems that the fix for #310 did not end up in the 0.10.6 version for Mac.

closed 2024-03-11 by prasimix